ccrisan/motionpie #378: Does this work with the zero?

nspireing, 2016-02-13:
I know it does not have the native connector but running a usb cam that works on another pi through a hub?

i tried using the Pi 1 build and it does not boot. i have a usb ethernet that works on my other pi's because i know it needs network to work correctly.

ccrisan, 2016-02-13:
No idea. I couldn't get my hands on a pi zero yet.

Ethocreeper, 2016-04-27:
you can try connecting a external network adapter or a WiFi dangle with raspi 2 image

nspireing, 2016-05-03:
i tried the motion eye for pi2 and all i get is a multicolored screen. like
a big under power square.

Ethocreeper, 2016-05-09:
that screen is when you boot up. it means your pi doesnt have enauf power to boot up.

maybe the usb Ethernet is drowning too much power try conecting your pi to a **powered **usb hub and connect the external network card in there.

if the network was the problem your rasberry will restart untill it detects network

ccrisan, 2016-05-09:
@nspireing Pi Zero has the same CPU as Pi1. Please use motionEyeOS (not motionPie!) for the first generation of the PIs: https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices#raspberry-pi-a-b-a-b-zero-variants.

Ethocreeper, 2016-05-13:
is your problem solved?

nspireing, 2016-05-13:
No i tried on two pi's it loads than fails. Im using a powered hub and a
usb to nic that i use often on the pi's