ccrisan/motionpie #380: Connection lost randomly

MaxTPS, 2016-02-29:
Hi, 

I am facing an weird issue on raspberry 2 with the latest release. 
Formating SD card ok. 
Imaging with last motionpie version ok
configuring thru web interface ok 
and then... at some time, the connection with the cam is lost.
no streaming, no recording, no access thru web or putty
but the ping works. 
Reformating, reimaging, reconfiguring works for a while and then the same issue appears after few hours. 

Any clue on this ?

MaxTPS, 2016-02-29:
[image]
here's a printscreen of putty connection refused, web interface KO and ping command OK. 
Hope it helps

ccrisan, 2016-02-29:
Could you please attach your log files?

MaxTPS, 2016-03-03:
Log files attached.

Kind regards,

ccrisan, 2016-03-03:
Please don't reply to the email, but use github instead.

MaxTPS, 2016-03-03:
BOOT.LOG 

---- booting motionEyeOS 20160221----
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Creating required data files: done
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: dhcp
 * Setting current date using http: Tue Mar 1 08:46:21 UTC 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.10/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1
---- shutting down motionEyeOS 20160221 ----
 * Stopping motioneye: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping brownout detector: done
 * Stopping watchdog: done
 * Stopping syslogd: done
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Tue Mar 1 09:50:47 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Tue Mar 1 09:56:35 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Tue Mar 1 09:58:17 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Tue Mar 1 10:35:20 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- shutting down motionEyeOS 20160221 ----
 * Stopping motioneye: failed
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping brownout detector: done
 * Stopping watchdog: done
 * Stopping syslogd: done
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Tue Mar 1 11:17:06 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- booting motionEyeOS 20160221----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.101/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Thu Mar 3 10:03:44 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.101/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8

MaxTPS, 2016-03-03:
[logs.zip](https://github.com/ccrisan/motionpie/files/156968/logs.zip)

MaxTPS, 2016-03-03:
there you go :-)

ccrisan, 2016-03-03:
Have you overclocked your PI?

ccrisan, 2016-03-03:
There's also the following part of your messages log:

```
Mar  1 11:23:57 meye-39438f06 user.notice motioneye: not responding for 120 seconds, rebooting
Mar  3 09:59:41 meye-39438f06 user.notice ethernet: disconnected
Mar  3 09:59:46 meye-39438f06 user.notice ethernet: disconnected
Mar  3 10:02:52 meye-39438f06 user.notice ethernet: disconnected
Mar  3 10:02:57 meye-39438f06 user.notice ethernet: disconnected
Mar  3 10:03:02 meye-39438f06 user.notice ethernet: disconnected
Mar  3 10:03:07 meye-39438f06 user.notice ethernet: disconnected
Mar  3 10:03:12 meye-39438f06 user.notice ethernet: disconnected for 20 seconds, rebooting

```

which indicates two things: the motioneye server somehow got frozen (a buggy USB camera?) and the ethernet connection is lost (check your cable/connector).

MaxTPS, 2016-03-04:
Today, i will change the utp cable and see if it's going better. i'll let you know.
yes the pi is overclocked (pi2 mounted with a csi camera)

ccrisan, 2016-03-04:
You could try disabling (or reducing) overclocking, as the dmesg file shows some instruction errors that usually appear when overheating.

MaxTPS, 2016-03-16:
hello and sorry for the late answer but i've been very busy making different tests, snatching my hairs, shouting out loud in the street, but now... i'm ok. :-) and i found what caused the issue. 

I tried these: 
Reduce overclocking
Change IP
Replace cable
Change camera
Change pi board

after all this, I figured out that the synology where the records are stored experienced some network disconnection. 
And that's the point. 
If the motionpieos is configured to store motions in the network and that device is not connected, the streaming won't appear in the Navigator. 
The camera still working, you can ping it but you can't Stream it unless you reconnect your recording device. 

hope it helps if someone is facing the problem again... 

Cheers.

MaxTPS, 2016-05-26:
hi everyone, 
sorry for digging up this post but i'm experiencing disconnection...again.... 
that's weird because the camera is working for a while (24-48h for example) and for no reason, the streaming is not reachable though the Navigator. 
Weird again, i can ping the pi but no access to it. 

here are the logs, the last disconnection was 2016-05-26 at 00h17 then retrieved few second after.
most of the time, there's 2 or 3 "disconnection-retrieve" and then it disconnects without retrieving until I unplug the pi to reboot it. 

have you ever experienced that ? 
[logs.zip](https://github.com/ccrisan/motionpie/files/284092/logs.zip)

ccrisan, 2016-05-26:
It's hard to say what the cause is. However I recommend switching to [motionEyeOS](https://github.com/ccrisan/motioneyeos) as it's supposed to be a bit more stable.

MaxTPS, 2016-05-26:
The thing is, i've tried with motionpie AND motioneyeOS

I was thinking, it happens only with outdoor camera where the pi's are completely enclosed in an enclosure without cooling. 
 
would you think overheat be the cause ?

ccrisan, 2016-05-26:
Judging by the logs and by the fact that motionEyeOS does the same, it's probably a hardware issue. It could be anything, including overheating or insufficient power. It's up to you to identify the problem. I couldn't help you with that.