ccrisan/motionpie #381: Cannot connect or view on anything unless logged in on PC

Lo42894, 2016-03-03:
Ok, I will do my best to describe this. I cannot stream video from internally or externally with my network unless I have a pc logged into the management page. The second I log into it via a browser at the main address for the management console, I can now stream on any other device to the camera address.

Situation:
Computer logged into 192.168.1.240
Can  view at 192.168.1.240:8081 on cell phone, or another pc and outside of network at external IP and port (port forward in router is working correctly)

With out computer however, I direct my device or pc to the address http://192.168.1.240:8081 and I cannot connect on any device in any location, local or external.

Help?

ccrisan, 2016-03-03:
Please switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos) if you haven't done it already and, in Expert Settings, turn the Motiion Keep-alive option on.

Lo42894, 2016-03-03:
Okay I am just using motion pie I think (https://github.com/ccrisan/motionPie/releases). So I guess I will start by updating my OS, ow after that, do I still have to use that setting? I thought it was part of the notification if it detected motion, but I assume that it is keeping the service alive regardless of motion?

Lo42894, 2016-03-04:
Thanks for your help all is well!

ccrisan, 2016-03-04:
That's great to hear.

closed 2016-03-04 by ccrisan