ccrisan/motionpie #385: Error when adding new cam

boarder01, 2016-03-14:
Hello,
My config.: 2 Raspberry 2 with one cam on each, 1 raspberry 2 without cam (server). with the same version of motioneye: 20160306.
When I try to add a remote motioneye cam to the raspberry server, I can see the cam. But after I clicked on ok, I have an error message :"An error occured, refreshing is recommended". An then I have always this message when I logg me on motioneye. The content of the motion log:

[motioneye]    ERROR: expected string or buffer
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py",
                                                                                 line 1401, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", 
                                                                                 line 1603, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", 
                                                                                 line 235, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", 
                                                                                 line 158, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", 
                                                                                 line 692, in add_camera
  File "/usr/lib/python2.7/site-packages/motioneye/remote.py", 
                                                                                 line 171, in get_config
  File "/usr/lib/python2.7/site-packages/motioneye/remote.py", 
                                                                                line 50, in _make_request
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 595, in compute_signature
TypeError: expected string or buffer
2016-03-11 09:47:11: 
[motioneye]    ERROR: 500 POST /config/add/?_=1457689631548&_username=admin&_signature=f816314b836d365eec11906a7e9f3850bea23d37 (153.109.46.1) 20.22ms
2016-03-11 09:47:52: 
[motioneye]    ERROR: failed to list remote cameras on http://153.109.56.20: authentication failed
2016-03-11 09:47:57:

ccrisan, 2016-03-14:
Are you sure your PI has a public IP address? Have you set up port forwarding on port 80 or so? If yes, I strongly recommend avoiding that. I'll try to reproduce the issue nevertheless.

boarder01, 2016-03-14:
Yes I have public IP. I can access to each raspberry to setup an see the cam with a web browser.

boarder01, 2016-03-14:
You can try to reach my camera:
http://153.109.56.20/ 

user: admin
pass: 161616

boarder01, 2016-03-14:
oups http://153.109.56.18

ccrisan, 2016-03-14:
Alright I am looking into it right now.

ccrisan, 2016-03-14:
I found the bug. A quick workaround would to be to change the password to something that's not a number (i.e. add a character to it).

boarder01, 2016-03-14:
I try it and all its ok. Thanks

closed 2016-03-14 by boarder01