ccrisan/motionpie #387: USB File Storage

BrianDurward, 2016-03-22:
Hi Crisan
I'm a Linux novice and trying to use your excellent system with a bird box camera.
I'm getting stuck with finding stored files when I use a USB file stick.
I select the drive from the file storage options but then can't access the drive using Filezilla.
I suspect i'm making a simple error.
Your help would be most welcome.

ccrisan, 2016-03-22:
First of all, please switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos). A screenshot of what you see in Filezilla would be useful.

BrianDurward, 2016-03-22:
Hi Calin

Many thanks for a fast response.
I have attached a screenshot of Filezilla.
I have also included a screen shot of the file settings in motionEyeOS.

Cheers

Brian

ccrisan, 2016-03-22:
Please attach the screenshots on github.

BrianDurward, 2016-03-22:
Uploaded them to github.

Brian