ccrisan/motionpie #392: RaspberryPi3 SSH host key problem

cappellen, 2016-04-18:
Thanks for the fantastic piece of software.

I have motioneyeos installed on a raspberry pi3 with the raspbarrypi3 version and it runs very smooth. However when I try to connect to the pi through SSH with Putty I receive a Putty fatal error "Couldn't agree a hsot key algorithm (available:ecdsa-sha2-nistp256,ssh-ed25519). The error comes from Putty, but it probably has to do with the SSH server of morioneyeos. Until now I was not able to solve this problem. Is it something with morioneyeos or with Putty. Other motioneyeos distributions I use and have used, didn't have this problem (on raspberry pi2 and 1).

ccrisan, 2016-04-18:
It might be a matter of first boot interruption (i.e. the ssh keys weren't correctly saved to the data partition). Please try reinstalling the OS image from scratch and do not poweroff or reboot the unit at first boot for at least say 10 minutes.

cappellen, 2016-04-21:
Thank you Calin it solved the problem.

closed 2016-04-21 by cappellen