ccrisan/motionpie #394: RPI 3 WIFI

JacoLouw, 2016-04-26:
Hi guys

Don't know what I am doing wrong, but I cannot connect the onboard wifi from the rpi 3. Always get wpa_supplicant failed (brcmfmac)
wireless network: no link

Don't know what else I can try....
Checked all the comments: WPA2 Personal, no special password etc.... also tried different power sources.

Please help! 

[boot (1).txt](https://github.com/ccrisan/motionpie/files/237571/boot.1.txt)
[dmesg (1).txt](https://github.com/ccrisan/motionpie/files/237573/dmesg.1.txt)
[messages.txt](https://github.com/ccrisan/motionpie/files/237574/messages.txt)

ccrisan, 2016-04-27:
If you're absolutely sure you entered the correct settings for your wpa-enabled network, then only two ideas come to my mind:

 * make sure the signal is good enough
 * make sure the setup works with Raspbian

JacoLouw, 2016-04-27:
Hi

Rpi is next to the router
Even changed password to 12345678
Changed ssid

Is there a issue with upper/lowercase ssid? Have a rpi2 with rp wifi Dongle which works on Raspbian. Tried this Dongle also on the rpi3 with no luck.

Ethocreeper, 2016-04-27:
maybe is case sensitive like Linux (Raspbian) commands

JacoLouw, 2016-04-27:
Hey guys

Thanks for your feedback.

Found the following:
1. The GUI saves the PSK in /data/etc/wpa_supplicant.conf with quotes (""). This cases "wpa_supplicant failed (brcmfmac)" at boot. Removing the quotes wifi works!
2. Making ANY changes in the GUI and clicking apply, updates /data/etc/wpa_supplicant.conf again adding another psk entry with quotes.
3. Disabling WIFI in the GUI and manually changing /data/etc/wpa_supplicant.conf ==> psk is not touched, but key_mgnt is set to NONE.

Can I stop the GUI from making changes to /data/etc/wpa_supplicant.conf ?
chattr +I is not supported

Thanks!

ccrisan, 2016-04-27:
@JacoLouw  that's weird. I'll have to look into it. Anyway, if you want the UI to no longer touch your wpa_supplicant.conf, try editing `/data/etc/motioneye.conf` and setting `wpa_supplicant_conf` to either `false` or to some random file (e.g. `/data/etc/wpa_supplicant_unused.conf`). This is untested but should work. Keep me posted.

JacoLouw, 2016-04-27:
Excellent! Did the trick! 
Thanks for all your support, appreciate it!

JacoLouw, 2016-04-27:
Forgot to ask: Is there a way to monitor or read the cpu temp?
Thanks!

ccrisan, 2016-04-27:
Nope. At least not an "official" one. And there won't be one in the near future.

JacoLouw, 2016-04-28:
That's a pity. 

I am planning on using a MoPi to run the pi  on batteries - will I be able to install the software for it on Motioneye?

ccrisan, 2016-04-28:
You can install anything you want in Raspbian :) I think that's your best option.