ccrisan/motionpie #396: Raspi 2 + Camera v2.1 Problem

jdaehler, 2016-05-05:
Hi, motionpie says "(no cameras)"!? Is the new cam not working with the current version?

Kayak83, 2016-05-11:
It's working on my Pi3. Not much else I can offer there unfortunately.

invisan, 2016-05-14:
Hi jdaehler, 

looks like its not only a Problem with the PI2 the PI Model B+ has the same Problem.
When trying modprobe bcm2835-v4l2 i get the message below:

modprobe: can't load module bcm2835-v4l2 (kernel/drivers/media/platform/bcm2835/ bcm2835-v4l2.ko): Operation not permitted

when looking at the output of dmesg i see this:

bcm2835-v4l2: error -1 while loading driver

Normally according to this thread:

https://github.com/ccrisan/motionpie/issues/30

That should be due to a HW/Camera fault BUT it is working fine with Raspbian so i guess it might be a Kernel issue.

invisan, 2016-05-14:
Gonna update this. 

Try to use https://github.com/ccrisan/motioneyeos/wiki instead. Works fine for me with my RPI and the RPI Cam v2